PaulStoffregen/MotionCal #8: Motion Cal is not displaying data on platform

kswain1, 2017-05-06:
So currently I am using a flora board on Arduino with motionCal software. I worked through this tutorial here [https://learn.adafruit.com/ahrs-for-adafruits-9-dof-10-dof-breakout/magnetometer-calibration](Adafruit Tutorial). I currently have the platform at the point where I can connect to it to serially( or so it seems), but for some reason there is no data being displaying on application once the flora board is booted up. This is how my code looks as sending the data to the motional app. 
`  // Print the sensor data
  Serial.print("Raw:");
  Serial.print(accel.raw.x);
  Serial.print(',');
  Serial.print(accel.raw.y);
  Serial.print(',');
  Serial.print(accel.raw.z);
  Serial.print(',');
  Serial.print(gyro.raw.x);
  Serial.print(',');
  Serial.print(gyro.raw.y);
  Serial.print(',');
  Serial.print(gyro.raw.z);
  Serial.print(',');
  Serial.print(mag.raw.x);
  Serial.print(',');
  Serial.print(mag.raw.y);
  Serial.print(',');
  Serial.print(mag.raw.z);
  Serial.println();  `

PaulStoffregen, 2017-05-07:
Sorry, I can't provide support for Flora.

Please ask on Adafruit's forum.

closed 2017-05-07 by PaulStoffregen

kswain1, 2017-05-07:
It's not about the flora board. It's about us Thur the correct way to send
data to the your calibration tool.

kswain1, 2017-05-08:
Hey Paul,

This was an easy problem to fix. You could have easily assisted me with this problem. I just had to change the data type of the raw sensor data to int, and unfortunately I had as a string data type when I was going through the error. I found this out by looking at your source code!

There are 1000's of other people using your open source library with alternative sensors by the way, and this would be very helpful for them to know how this works!
  ` lsm.read();

  // Print the sensor data
  Serial.print("Raw:");
  Serial.print((int)lsm.accelData.x);
  Serial.print(',');
  Serial.print((int)lsm.accelData.y);
  Serial.print(',');
  Serial.print((int)lsm.accelData.z);
  Serial.print(',');
  Serial.print((int)lsm.gyroData.x);
  Serial.print(',');
  Serial.print((int)lsm.gyroData.y);
  Serial.print(',');
  Serial.print((int)lsm.gyroData.z);
  Serial.print(',');
  Serial.print((int)lsm.magData.x);
  Serial.print(',');
  Serial.print((int)lsm.magData.y);
  Serial.print(',');
  Serial.print((int)lsm.magData.z);
  Serial.println();  `

slowtoaccept, 2018-03-16:
Bravo kswain1. Paul's code is very much appreciated, BUT, w/o some hand-holding its usefulness is limited.

cypyn, 2018-03-29:
Just to add to this discussion. It was necessary for me to scale the FLORA output significantly for the LSMD9S0 chip to calibrate with MotionCal using the Adafruit_LSM9DS0_Library. The setup above for the FLORA would not calibrate for me. Here is what worked:

sensors_event_t accel, mag, gyro, temp;
  lsm.getEvent(&accel, &mag, &gyro, &temp); 
  Serial.print("Raw:");
  Serial.print((int)(accel.acceleration.x*1000));
  Serial.print(',');
  Serial.print((int)(accel.acceleration.y*1000));
  Serial.print(',');
  Serial.print((int)(accel.acceleration.z*1000));
  Serial.print(',');
  Serial.print((int)(gyro.gyro.x*1000));
  Serial.print(',');
  Serial.print((int)(gyro.gyro.y*1000));
  Serial.print(',');
  Serial.print((int)(gyro.gyro.z*1000));
  Serial.print(',');
  Serial.print((int)(mag.magnetic.x*1000));
  Serial.print(',');
  Serial.print((int)(mag.magnetic.y*1000));
  Serial.print(',');
  Serial.print((int)(mag.magnetic.z*1000));
  Serial.println();

PaulStoffregen, 2018-03-29:
I recommend you submit this to Adafruit.

pcmonahan, 2019-01-23:
Greetings All,

Instead of a direct hardware connection with realtime display,  If I wanted to format data that has been previously captured from my hardware and stored as CSV files, then send the CSV file to MotionCal using a terminal emulator, what would the correct formatting be for the CSV file ?

Like this?
Raw:0,0,0,0,0,0,123,456,789 

Do I need to add a CR or LF?

At this time, I am only interested in magnetometer assessment.

Thanks

RachelDunwoody, 2019-02-07:
How did you figure out the scaling factor? I have a different magnetometer (BMM150 Seeed Studio 3-axis Compass V2.0) - I have very few red points showing up and very small so unsure how to fix this - any suggestions appreciated!

duckman1961, 2019-10-13:
If you have an adafruit NXP, you can check the serial data size, then put in a scale factor to match the data that works. My NXP has a range of approx+-800, so l need to put a *10 to get the raw data l saw in serial monitor for a MPU9255 to work in MotionCal.
Also found after a lot of hair pulling because l couldn't get the red dots to show/work in MotionCal l had a space between the ("  , " ).
Doesn't matter whether you use (",") or (',') as long as no space between the " or '  .
Your serial Monitor data must look like below to work in MotionCal.
Raw:0,0,0,0,0,0,168,-95,-429
NOT like this example below, which has 1 float # and the scale is wrong.
Raw:0,0,0,0,0,0,2839.08,-1647,-7235

You have to have ("Raw: ") at the start and 9 separate bits of data in the serial print for it to work, as it uses the last 3 (mag) to operate.
It also uses data type (int) not real #'s, so that has to be added to correct any data you input.

But l suggest you use Magneto12 to do the calibration, as it will give you the correct scaling because you use the initial data of the IMU/ARHS so will have the correctly scaled bias, the correction matrix will be close/same.
I  wasn't able to get the accel/gyro data to display, but you don't want to hard code that in anyway, you want it to sample on every start up and use this correction bias when running.