PaulStoffregen/MotionCal #9: In RawData you assign the Y axis twice to the accelerometer variable but never the Z axis

matteofranceschini, 2017-08-27:
https://github.com/PaulStoffregen/MotionCal/blob/d579c0dcf5f010670d6a42fe9540a37a18269136/rawdata.c#L189

e27182, 2018-01-21:
Paul, would it be possible to update and recompile MotionCal at least with this fix? It could significantly influence calibration process.

PaulStoffregen, 2018-01-23:
Fixed.  https://github.com/PaulStoffregen/MotionCal/commit/f8f279563f9c0c7af34b00fabbf20638c4ab7982

My Linux build system is currently broken for GLX.  Need to get that fixed or use another Linux machine for testing, before I can rebuild all the binaries.  Remind me next week if this falls through the cracks & they're not updated...

PaulStoffregen, 2018-01-23:
Oh, looks like I need an update on the Mac build too, for Apple's new Gatekeeper signatures.  Grrrrr.....

closed 2018-01-23 by PaulStoffregen

PaulStoffregen, 2018-01-23:
Compiled binaries updated.