PaulStoffregen/MotionCal #10: Compilation problems on ubuntu 18

altineller, 2019-08-30:
Hello,

I am trying to compile this program in an effort to use it with ros, without requiring a special firmware for the imu. 

I downloaded the source code, and initially make complained about wxwidgets and I installed all the libraries to fix them, but I now get an error that I can not fix:

`g++ -s -O2 -Wall -DLINUX  -o MotionCal gui.o portlist.o images.o visualize.o serialdata.o rawdata.o magcal.o matrix.o fusion.o quality.o mahony.o `wx-config --libs all,opengl`
/usr/bin/ld: visualize.o: undefined reference to symbol 'glMaterialfv'
//usr/lib/x86_64-linux-gnu/libGL.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Makefile:69: recipe for target 'MotionCal' failed
make: *** [MotionCal] Error 1
`

tavvfiq, 2019-09-18:
yes i have this issue too. how to fix it?

PaulStoffregen, 2019-09-18:
https://forum.pjrc.com/threads/57378-Cannot-make-MotionCal-for-Linux

altineller, 2019-09-27:
> https://forum.pjrc.com/threads/57378-Cannot-make-MotionCal-for-Linux

It is fixed now, and I am able to compile. Here is what I had to do (in ubuntu 18)

1. Modified the make file to have: WXCONFIG = /usr/bin/wx-config
2. g++ -s -O2 -Wall -DLINUX  -o MotionCal gui.o portlist.o images.o visualize.o serialdata.o rawdata.o magcal.o matrix.o fusion.o quality.o mahony.o `/usr/bin/wx-config --libs all,opengl` -lGL -lGLU

(I had to manually add the -lGL -lGLU to the last g++ compilation command)

It does compile and run at the moment.

closed 2019-09-27 by altineller