PaulStoffregen/MotionCal #11: Manually specify the port

WayneKeenan, 2020-05-25:
The macOS port list function doesn't find my FTDI based Sparkfun ESP32 Thing USB CDC/serial port.  Obviously it would be nice to fix the underlying enum issue, but instead of any potential wack'a'mole-ing...

It would be helpful to be able to manually specify a port (as a command line arg or in the GUI)  instead of only allowing selection from a pre-populated list.

matteocarnelos, 2022-04-12:
Hey @WayneKeenan, I solved this issue by manually building the latest version of the application hosted in this repo:
```
$ brew install wxwidgets
$ git clone https://github.com/PaulStoffregen/MotionCal
$ cd MotionCal
[comment `OS = LINUX` and uncomment `OS = MACOSX_CLANG` in `Makefile`]
$ make
```
And finally you can open `MotionCal.app`.

Here's the version I built: [MotionCal.app.zip](https://github.com/PaulStoffregen/MotionCal/files/8474234/MotionCal.app.zip)

WayneKeenan, 2022-04-12:
Thanks. I'll give it a go at some point, just that I can't remember what I was using this for now lol.  (Will have to check notes / projects)

IanBUK, 2022-07-16:
> Hey @WayneKeenan, I solved this issue by manually building the latest version of the application hosted in this repo:
>
> ```
> $ brew install wxwidgets
> $ git clone [...]

On Monterey, M1 MacBook Air, I get the following:
```fatal: Could not resolve HEAD to a revision
Warning: No available formula with the name "wxwidgets".
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.```

IeiuniumLux, 2022-12-30:
I get the same errors like @IanBUK, is OSX Monterey supported?

matteocarnelos, 2023-01-02:
@IeiuniumLux it should be supported (see [here](https://formulae.brew.sh/formula/wxwidgets)). It seems that the error is related to the homebrew installation, [this StackOverflow question](https://stackoverflow.com/questions/41113448/homebrew-error-no-formulae-found-in-taps) might help.

IeiuniumLux, 2023-01-03:
@matteocarnelos,

I have wxwidgets installed, but the problem is that it doesn't list the port for the device connected to the Mac.

[image]

SouthernAtHeart, 2023-03-18:
I too am trying to use the MotionCal software or my MacBook, and it doesn't list any ports, as in the above photo...

mhaberler, 2023-04-12:
I just tried on a M1 Macbook
brew install wxwidgets

add -arch arm64  to CFLAGS and build for OS = MACOSX_CLANG

make
open MotionCal.app

works for me!

````
diff --git a/Makefile b/Makefile
index 83a8d7a..a41dbb8 100644
--- a/Makefile
+++ b/Makefile
@@ -1,6 +1,6 @@
-OS = LINUX
+#OS = LINUX
 #OS = MACOSX
-#OS = MACOSX_CLANG
+OS = MACOSX_CLANG
 #OS = WINDOWS

 ifeq ($(OS), LINUX)
@@ -32,10 +32,10 @@ else ifeq ($(OS), MACOSX_CLANG)
 ALL = MotionCal.app
 CC = /usr/bin/clang
 CXX = /usr/bin/clang++
-CFLAGS = -O2 -Wall -DMACOSX
+CFLAGS = -O2 -Wall -DMACOSX -arch arm64
 WXCONFIG = wx-config
 WXFLAGS = `$(WXCONFIG) --cppflags`
-CXXFLAGS = $(CFLAGS) `$(WXCONFIG) --cppflags`
+CXXFLAGS = $(CFLAGS) `$(WXCONFIG) --cppflags`
 SFLAG =
 CLILIBS = -lglut -lGLU -lGL -lm
 VERSION = 0.01
````

[image]

virtyaluk, 2023-08-14:
The provided MacOS executable doesn't work for me. Building from the source also didn't help to fix the empty list of ports. Managed to get a working port list on the latest Windows 10, but the app doesn't seem to listen to the data coming from MCU at all. Had to consider an alternative and most importantly working solution - MagCal.

UnexpectedMaker, 2023-12-09:
I know it's been a while...  I also just had this issue on macOS 14.1.2 on an M2 Max - but I followed @mhaberler 's steps and I re-compiled (with a few errors in the output) but it worked, and I see the ports in the list and I am able to use the tool.

PipoPL, 2024-02-19:
I had the same issue but just built the app with  @mhaberler  instructions and it worked !  MacOs Monterrey 12.1. 
thank you !

josephlarralde, 2024-04-10:
Same here on Ventura 13.5 on a Apple M2 Max, thanks @mhaberler !

chunghay, 2024-10-15:
I had the same issue on macOS Sonoma 14.6.1 on a MacBook Pro, and @mhaberler's instructions worked!

In case others benefit from a more detailed explanation of the instructions, I listed them below:

Delete any earlier instances of MotionCal.app, e.g. one that was created from the .dmg file

Then in the terminal, run this to get the software for displaying GUIs:
```
brew install wxwidgets
```

Continue in the terminal with these commands to get the source code:
```
cd /Applications
git clone https://github.com/PaulStoffregen/MotionCal.git
```

In the terminal, go to the created folder:
```
cd MotionCal
open .
```

You should see the folder "MotionCal" open in a separate window. Find the file "Makefile" and open it in a text editor. Make the following changes (to get what's in bold) and save the file:

- Comment out the Linus OS row…  
  - OS = LINUX  --> **#OS = LINUX**
- Uncomment the MACOSX_CLANG option row… 
  - #OS = MACOSX_CLANG --> **OS = MACOSX_CLANG**
- Add “-arch arm64” at the end of the MACOS CFLAGS row… 
  - CFLAGS = -O2 -Wall -DMACOSX --> **CFLAGS = -O2 -Wall -DMACOSX -arch arm64**

Then in the terminal (you should be in /Applications/MotionCal), compile the code:
`make`

Now you can open the created app:
`open MotionCal.app`

flyboyhunt, 2024-11-22:
Hey. If anyone has some input on how to fix this I would appreciate it. Followed instructions until MAKE terminal command. This proceeds the following error.

I have already have wxwidgets installed

any suggestions?

/usr/bin/clang++ -O2 -Wall -DMACOSX -arch arm64 `wx-config --cppflags`   -c -o gui.o gui.cpp
In file included from gui.cpp:1:
In file included from ./gui.h:4:
In file included from /usr/local/include/wx-3.2/wx/wx.h:24:
In file included from /usr/local/include/wx-3.2/wx/event.h:22:
In file included from /usr/local/include/wx-3.2/wx/cursor.h:75:
/usr/local/include/wx-3.2/wx/utils.h:749:35: warning: default member initializer for non-static data member is a C++11 extension [-Wc++11-extensions]
    wxEventLoop* m_modalEventLoop = NULL;
                                  ^
1 warning generated.
/usr/bin/clang++ -O2 -Wall -DMACOSX -arch arm64 `wx-config --cppflags`   -c -o portlist.o portlist.cpp
In file included from portlist.cpp:5:
In file included from ./gui.h:4:
In file included from /usr/local/include/wx-3.2/wx/wx.h:24:
In file included from /usr/local/include/wx-3.2/wx/event.h:22:
In file included from /usr/local/include/wx-3.2/wx/cursor.h:75:
/usr/local/include/wx-3.2/wx/utils.h:749:35: warning: default member initializer for non-static data member is a C++11 extension [-Wc++11-extensions]
    wxEventLoop* m_modalEventLoop = NULL;
                                  ^
portlist.cpp:217:6: warning: 'IOMasterPort' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        if (IOMasterPort(NULL, &masterPort) != KERN_SUCCESS) return list;
            ^~~~~~~~~~~~
            kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:143:1: note: 'IOMasterPort' has been explicitly marked deprecated here
IOMasterPort( mach_port_t       bootstrapPort,
^
portlist.cpp:217:19: warning: implicit conversion of NULL constant to 'mach_port_t' (aka 'unsigned int') [-Wnull-conversion]
        if (IOMasterPort(NULL, &masterPort) != KERN_SUCCESS) return list;
            ~~~~~~~~~~~~ ^~~~
                         0
3 warnings generated.
perl png2c.pl checkgreen.png checkempty.png checkemptygray.png > images.cpp
/usr/bin/clang++ -O2 -Wall -DMACOSX -arch arm64 `wx-config --cppflags`   -c -o images.o images.cpp
/usr/bin/clang -O2 -Wall -DMACOSX -arch arm64   -c -o visualize.o visualize.c
visualize.c:67:2: warning: 'glClear' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
        glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);
        ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2394:13: note: 'glClear' has been explicitly marked deprecated here
extern void glClear (GLbitfield mask) OPENGL_DEPRECATED(10.0, 10.14);
            ^
visualize.c:68:2: warning: 'glColor3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
        glColor3f(1, 0, 0);     // set current color to red
        ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2405:13: note: 'glColor3f' has been explicitly marked deprecated here
extern void glColor3f (GLfloat red, GLfloat green, GLfloat blue) OPENGL_DEPRECATED(10.0, 10.14);
            ^
visualize.c:69:2: warning: 'glLoadIdentity' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
        glLoadIdentity();
        ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2571:13: note: 'glLoadIdentity' has been explicitly marked deprecated here
extern void glLoadIdentity (void) OPENGL_DEPRECATED(10.0, 10.14);
            ^
visualize.c:112:5: warning: 'glPushMatrix' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                                glPushMatrix();
                                ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2625:13: note: 'glPushMatrix' has been explicitly marked deprecated here
extern void glPushMatrix (void) OPENGL_DEPRECATED(10.0, 10.14);
            ^
visualize.c:116:5: warning: 'glTranslatef' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                                glTranslatef(
                                ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2729:13: note: 'glTranslatef' has been explicitly marked deprecated here
extern void glTranslatef (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^
[... 1879 lines cut ...]
  "_wxStaticBitmapNameStr", referenced from:
      MyFrame::MyFrame(wxWindow*, int, wxString const&, wxPoint const&, wxSize const&, long) in gui.o
  "_wxStaticTextNameStr", referenced from:
      MyFrame::MyFrame(wxWindow*, int, wxString const&, wxPoint const&, wxSize const&, long) in gui.o
  "_wxTheAssertHandler", referenced from:
[... cut ...]
      wxEventFunctorMethod<wxEventTypeTag<wxFocusEvent>, wxNavigationEnabled<wxWindow>, wxFocusEvent, wxNavigationEnabled<wxWindow> >::operator()(wxEvtHandler*, wxEvent&) in gui.o
      wxEventFunctorMethod<wxEventTypeTag<wxChildFocusEvent>, wxNavigationEnabled<wxWindow>, wxChildFocusEvent, wxNavigationEnabled<wxWindow> >::operator()(wxEvtHandler*, wxEvent&) in gui.o
      ...
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [MotionCal] Error 1

srgg, 2025-01-02:
Thanks to @mhaberler, I was able to build and run Motion Cal, but with Adafruit reverse TFYT feather connected via type-c, I can not choose '/dev/cu.usbmodem2101', it is shown in the dropdown list, but as soon as I tried to select it, it selects '(none)' :(

[image]

2e0wkw, 2025-01-13:
I cannot make MotionCal on my M1 MacMini running MacOs 15.1.1.
I get:
ld: symbol(s) not found for architecture arm64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [MotionCal] Error 1

Eric

naimmas, 2025-06-27:
@2e0wkw me too I got the same error. any suggestions?

IanBUK, 2025-06-27:
> @2e0wkw me too I got the same error. any suggestions?
> 

Take a look at the open pr -  no. 22. This is mine and works well on my M1 MacBook Air.

naimmas, 2025-06-27:
> Take a look at the open pr - no. 22. This is mine and works well on my M1 MacBook Air.

Thanks. I managed to build it on M2 Sequio as well, but I had  to use `arch -x86_64` with the following flags. I also built wxWidget from source.

```
diff --git a/Makefile b/Makefile
index 7db71bd..d33ad40 100644
--- a/Makefile
+++ b/Makefile
@@ -66,7 +66,7 @@ IMGS = checkgreen.png checkempty.png checkemptygray.png
 all: $(ALL)
 
 MotionCal: gui.o portlist.o images.o $(OBJS)
-       $(CXX) $(SFLAG) $(CFLAGS) $(LDFLAGS) -o $@ $^ `$(WXCONFIG) --libs all,opengl`
+       $(CXX) $(SFLAG) $(CFLAGS) $(LDFLAGS) -framework Cocoa -framework OpenGL -framework IOKit -o $@ $^ `$(WXCONFIG) --libs all,opengl `
 
 MotionCal.exe: resource.o gui.o portlist.o images.o $(OBJS)
        $(CXX) $(SFLAG) $(CFLAGS) $(LDFLAGS) -o $@ $^ `$(WXCONFIG) --libs all,opengl`
```

llamafilm, 2026-02-09:
The instructions from @chunghay worked for me, but I also had to remove the `pjrcmacsigntool` line.