PaulStoffregen/MotionCal #12: Data input range/scale documentation

apateluk, 2021-02-03:
Is the data input range and scale documented somewhere ? 

I'm trying to use an MPU6050 and Qmc5883L with Motioncal and the raw data being sent to Motioncal - but i'm not getting the expted output. I;ve looked at a few examples and see some of the data is being scaled on the MCU before sending to MotionCAL - So wondering what exactly MotionCal is expecting to see.

Thanks

drthegirlfriend, 2021-07-26:
I have the same questions...Though according to the adafruit comments in their code, it seems the software looks for 

Accel in 'raw' format 2^13 (8192) integers
Gyroscope in Degrees/s rounded to integers
Mag in microteslas * 10

Though I get the motioncal software to work when using 2^14 raw accel format and the gyroscope in degrees/s, and the mag multiplied by 10 I actually don't believe the values are particularly correct, because I'm having quite a difficult time getting real angles from the sensors itself (i'm using MPU9250)

drthegirlfriend, 2021-07-26:
I have found this in the [imuread header file](https://github.com/PaulStoffregen/MotionCal/blob/master/imuread.h):

```
// accelerometer sensor structure definition
#define G_PER_COUNT 0.0001220703125F  // = 1/8192
typedef struct
{
	float Gp[3];           // slow (typically 25Hz) averaged readings (g)
	float GpFast[3];       // fast (typically 200Hz) readings (g)
} AccelSensor_t;

// gyro sensor structure definition
#define DEG_PER_SEC_PER_COUNT 0.0625F  // = 1/16

// magnetometer sensor structure definition
#define UT_PER_COUNT 0.1F
typedef struct
{
	float Bc[3];           // slow (typically 25Hz) averaged calibrated readings (uT)
	float BcFast[3];       // fast (typically 200Hz) calibrated readings (uT)
} MagSensor_t;
```
so I assume, it assumes data as I mentioned, accel +/- 2^13 int, gyro degrees/s * 16, mag *10...though, a confirmation would be great.