PaulStoffregen/MotionCal #16: make error {make: pjrcmacsigntool: No such file or directory}

CodyWangYaohui, 2022-10-27:
Hi Paul, 

I am doing on a macOS and could not build the application properly. I've changed the OS to be macOS clang in the makefile and still could not make the project. It's always reporting pjrcmacsigntool is not found. The produced application could be opened but there are some problems with the port listening functionalities. When the correct port is chosen, it instantly changed to (none) with no errors rendered.

[image]

PaulStoffregen, 2022-10-28:
The build does not require pjrcmacsigntool.  Notice it says "Error 1 (ignored)".  This is only a script which adds a digital signature.  If you really want to digitally sign the app (won't solve the other problems), you'll need to get a code signing cert from Apple and use it to sign the app yourself.

CodyWangYaohui, 2022-10-28:
Hi Paul, thanks so much for your reply. Currently I can use the application, however the trace seems really random and 100% not a circle or sphere described in Adafruit instruction websites.[adafruit_magnetometer_calibration](https://learn.adafruit.com/adafruit-sensorlab-magnetometer-calibration/magnetic-calibration-with-motioncal)  I've tried plotting all three axis reading on jupyter notebook and attached here. Do you think it's the hardware malfunctioning? But I've actually tried two adafruit LSM6DSOX+LIS3MDL ICs, and they all report similar drawings.

Sorry to bother u again. I'm new to imu calibration and fusions, and I think you are the kind of experts that I should consult. Thanks so much again!
Cody
[image]
[image]
[image]

PaulStoffregen, 2022-10-28:
Ask Adafruit for support with their products!