PaulStoffregen/MotionCal #20: MotionCal not recognizing data from micro:bit v2.20

GillisICWE, 2025-03-04:
Hello,

I'm trying to use MotionCal with my micro:bit v2.20 to calibrate the built-in magnetometer. I know MotionCal is primarily designed for Teensy boards, but I was wondering if it could work with other hardware as long as the data stream follows the same format.

Here's my setup:

- Micro:bit v2.20
- MicroPython (via Mu editor)
- Serial communication via UART at 115200 baud
- MotionCal detects my micro:bit as a COM port on windows 11
- Data stream verified with Putty

> Raw:0,0,0,0,0,0,-18,33,-217
> Raw:0,0,0,0,0,0,58,-55,-148
> Raw:0,0,0,0,0,0,102,-195,-9
> Raw:0,0,0,0,0,0,138,-268,177
> Raw:0,0,0,0,0,0,87,-301,279

Delay in the code adjusted to 100 ms
This is the code running on the micro:bit:

```Python
from microbit import *
import time

uart.init(baudrate=115200)

while True:
    magnet_strength_x = compass.get_x()
    magnet_strength_y = compass.get_y()
    magnet_strength_z = compass.get_z()
    
    #new_byte = (uart.read(1))
    #if new_byte == None:
    #    continue
    raw = "Raw:0,0,0,0,0,0," + str(int(magnet_strength_x/100))+","+ str(int(magnet_strength_y/100))+","+ str(int(magnet_strength_z/100))+"\n"
    uart.write(raw.encode('utf-8'))
    #uni = "Uni:0,0,0,0,0,0," + str(magnet_strength_x/1000)+","+ str(magnet_strength_y/1000)+","+ str(magnet_strength_z/1000)+"\n"
    #uart.write(uni.encode('utf-8'))
    time.sleep_ms(100)
```
However, MotionCal shows no response. Has anyone successfully used MotionCal with a micro:bit?

Are there any specific formatting requirements for the serial data stream that MotionCal expects?

Thanks in advance!

IanBUK, 2025-04-24:
I'm not in front of my computer right now, but I had similar issues and it turned out to be my baud rate and line endings.

GillisICWE, 2025-04-25:
It was indeed the control end.

For reference the following code works.

```Python
from microbit import *
import time

uart.init(baudrate=115200)

while True:
    magnet_strength_x = compass.get_x()
    magnet_strength_y = compass.get_y()
    magnet_strength_z = compass.get_z()
    
    raw = "Raw:0,0,0,0,0,0," + str(int(magnet_strength_x/100))+","+ str(int(magnet_strength_y/100))+","+ str(int(magnet_strength_z/100))+"\r\n"
    uart.write(raw.encode('utf-8'))
    time.sleep_ms(100)
```

IanBUK, 2025-04-25:
I've got a fork where I'm adding a few things and will be making the baud rate etc. configurable.

tosa-no-onchan, 2025-05-05:
Hi.
My case is.
"Raw:0,0,0,0,0,0,-241,26,447"
"Raw:0,0,0,0,0,0,-246,28,442"
"Raw:0,0,0,0,0,0,-245,24,443"
"Raw:0,0,0,0,0,0,-245,24,443"
....
"Raw:0,0,0,0,0,0,mx,my,mz"
mx , my , mz : ICM-20948 Mageto Raw data
Out put Rate : 90[hz]
Serial Speed: 115200

SOC and IMU device.
ESP32 and ICM-20948
ESP32 Arduino SparkFun ICM-20948 Library.
Data code is not encode to utf-8. Perhaps Arduino base code.

PC: Ubuntu 24.04
Build MotionCal from source code.

When MotionCal start calibrate mageto data,
I rotate the IMU-Device any direction on the same position. 
MotionCal does work.

IanBUK, 2025-05-29:
My fork, at https://github.com/IanBUK/MotionCal, allows you to set the baud rate at runtime, and line endings. I've enabled it to work with a ' ' in the ascii package - my esp32 code was sending "Raw: " not "Raw:" and it took me longer than it should to understand why MotionCal wasn't using the package. #FacePalm

I've still got a 'noticeable' amount of work to do on it, to add something that's going to help my project, and I don't have access to a Windows or Linux machine to test on those platforms, but feel free to clone my fork and see if it does the good stuff for you.